MangoAutomation/ma-dashboards #10: Directive for computing sum over a time range on a point history

WillGeller, 2016-04-26:
Feature request for: http://localhost:8080/modules/dashboards/web/mdAdmin/#/dashboard/examples/basics/point-values

Can we make <ma-point-values> able to receive a data preset attribute? There will often be the circumstance where we don't want to us <ma-data-range-picker style="display: none;"> hack to get historical values over a time range.

In addition it would be nice to have a version that can incorporate the sum filter with last values so we don't have to do this:

<ma-point-values point="pointHistory" values="todaysValues" from="fromToday" to="toToday" rollup="LAST" rollup-interval="1 days">
{{todaysValues | sum:'value' | number:2}}

jazdw, 2016-05-03:
As discussed with Will, this is achieved via a ma-now directive and a using the moment filter. Will is going to add an example.

closed 2016-05-03 by jazdw

WillGeller, 2016-05-03:
Before I add an example, I have been playing around with the code and while it seems to work as expected returning the first and last from the time range, it is locking the page up and throwing an angular infinite digest loop.

Here is the code I am using in the play area:

`<ma-now update-interval="10 seconds" output="now"></ma-now>
Notify Jared of infinite digest bug error caused by the moment from

<div layout="row">
    <md-input-container flex="" class="no-errors-spacer">
        <label>Choose a point</label>
        <ma-point-list ng-model="myPoint"></ma-point-list>
    </md-input-container>
</div>

<ma-now update-interval="10 seconds" output="now"></ma-now>
<p>The time now as formatted string is {{now | moment:'format':'ll LTS'}}.</p>
{{ now }}
{{ happyFrom = (now|moment:'subtract':1:'days')}}
<p>The time from as formatted string is {{happyFrom | moment:'format':'ll LTS'}}.</p>

<ma-point-statistics point="myPoint" from="happyFrom" to="now" first-last="true" statistics="stats">
</ma-point-statistics>
	
<div>The delta is {{ stats[1].value - stats[0].value }}</div><br>

<div>From {{stats[0].timestamp | moment:'format':'ll LTS' }} to {{stats[1].timestamp |moment:'format':'ll LTS'}}</div><br>
<div>{{ stats}}</div>`

Digest error shown:

[https://docs.angularjs.org/error/$rootScope/infdig?p0=10&p1=%5B%5B%7B%22msg%22:%22fn:%20function%20(c,e,f,g)%7Bf%3Dd%26%26g%3Fg%5B0%5D:a(c,e,f,g);return%20b(f,c,e)%7D%22,%22newVal%22:%222016-05-02T15:57:52.145Z%22,%22oldVal%22:%222016-05-02T15:57:52.145Z%22%7D%5D,%5B%7B%22msg%22:%22fn:%20function%20(c,e,f,g)%7Bf%3Dd%26%26g%3Fg%5B0%5D:a(c,e,f,g);return%20b(f,c,e)%7D%22,%22newVal%22:%222016-05-02T15:57:52.145Z%22,%22oldVal%22:%222016-05-02T15:57:52.145Z%22%7D%5D,%5B%7B%22msg%22:%22fn:%20function%20(c,e,f,g)%7Bf%3Dd%26%26g%3Fg%5B0%5D:a(c,e,f,g);return%20b(f,c,e)%7D%22,%22newVal%22:%222016-05-02T15:57:52.145Z%22,%22oldVal%22:%222016-05-02T15:57:52.145Z%22%7D%5D,%5B%7B%22msg%22:%22fn:%20function%20(c,e,f,g)%7Bf%3Dd%26%26g%3Fg%5B0%5D:a(c,e,f,g);return%20b(f,c,e)%7D%22,%22newVal%22:%222016-05-02T15:57:52.145Z%22,%22oldVal%22:%222016-05-02T15:57:52.145Z%22%7D%5D,%5B%7B%22msg%22:%22fn:%20function%20(c,e,f,g)%7Bf%3Dd%26%26g%3Fg%5B0%5D:a(c,e,f,g);return%20b(f,c,e)%7D%22,%22newVal%22:%222016-05-02T15:57:52.145Z%22,%22oldVal%22:%222016-05-02T15:57:52.145Z%22%7D%5D%5D](url)

jazdw, 2016-05-04:
This occurs because every time a digest loop runs the moment filter returns a new object, angular detects that the end result of the expression is not `===` equal and runs another digest loop. This can be worked around by including  `| moment:'valueOf'` at the end of your filter expression so that angular ends up comparing an integer instead of an object. You also will not see the problem if you format the moment to a string since the strings will be `===` equal too.

In short, the following works.
````
<ma-now update-interval="10 seconds" output="now"></ma-now>

<div layout="row">
    <md-input-container flex class="no-errors-spacer">
        <label>Choose a point</label>
        <ma-point-list ng-model="myPoint"></ma-point-list>
    </md-input-container>
</div>

<p>The time now as formatted string is {{now | moment:'format':'ll LTS'}}.</p>
<p>The time from as formatted string is {{ now | moment:'subtract':1:'days' | moment:'format':'ll LTS'}}.</p>

<ma-point-statistics point="myPoint" from="now | moment:'subtract':1:'days' | moment:'valueOf'" to="now" first-last="true" statistics="stats">
</ma-point-statistics>

<pre>{{ stats | json }}</pre>
````

The only way to avoid this without converting to an integer or a string is to cache the result for every given input to the moment filter. I'll take a look at implementing this.

jazdw, 2016-05-04:
@WillGeller I've added caching to the moment and duration filters in ade79e596e2c1499f054183f3a2a95142da730a8. It should allow up to 10 moment filters to be chained before you get an infinite digest now.