MangoAutomation/ma-dashboards #9: Can we inject aria-label attr to menu to resolve warnings?

WillGeller, 2016-04-26:
Getting about 64 warnings from all the menu's md-button items and md-checkbox on the mdAdminTemplate.

`ARIA: Attribute " aria-label ", required for accessibility, is missing on node: <button class=​"menu-toggle-button md-button md-ink-ripple" type=​"button" ng-transclude ng-click=​"toggle()​">​…​</button>​
angular`

Can we inject the translation string as aria-label to clean up the error console?

jazdw, 2016-05-03:
@WillGeller I couldn't replicate the warnings from the menu links but it might have to do with load order. I was getting warnings on the login page. This should fix the warnings regardless.

closed 2016-05-03 by jazdw

WillGeller, 2016-05-03:
Warnings are now not appearing on mdAdmin;

can we copy the fix over to adminTemplate (the user starter kit-yellow version)?

jazdw, 2016-05-04:
Yep can do.

commit 39468630 (2016-05-03): Load translations up-front so they can be used by filter. Add
aria-labels to menu links. Closes #9