MangoAutomation/ma-dashboards #12: Angular vendor files .js.map error

WillGeller, 2016-05-07:
should we put in the .map files to avoid getting there errors in Safari when the console is open.

> Failed to load resource: the server responded with a status of 404 (Not Found)
http://localhost:8080/modules/dashboards/web/vendor/angular-animate/angular-animate.min.js.mapFailed to load resource: the server responded with a status of 404 (Not Found)
http://localhost:8080/modules/dashboards/web/vendor/angular/angular.min.js.mapFailed to load resource: the server responded with a status of 404 (Not Found)
http://localhost:8080/resources/angular-resource.min.js.mapFailed to load resource: the server responded with a status of 404 (Not Found)

Info: [http://stackoverflow.com/questions/18425841/angular-min-js-map-not-found-what-is-it-exactly](url)

terrypacker, 2016-05-08:
It would make the module bigger, that's why we don't package them with the core.  But since this module is going to be used by developers its worth considering a way to optionally include them so that developers lives are a little easier.

jazdw, 2016-05-10:
I know in Chrome you can turn off the loading of the .map files which will get rid of the console errors. but Terry raises a valid point, it might help developers working with Custom Dashboards. I'll take a look at including them.

jazdw, 2016-05-10:
I'm just going to leave it how it is, only the angular files have a .map file built for them and we are currently removing .min from the file extensions. I don't think its worth changing this, just disable the loading of the .map files or use Chrome if you can't.

closed 2016-05-10 by jazdw