MangoAutomation/ma-dashboards #13: <ma-point-query> way to prevent blank string loading all points in system

WillGeller, 2016-05-13:
Clearing out the search causes the page to attempt to load every point in the system into my statistics table. It causes the page to pretty much timeout and crash.

Is there a way to prevent blank strings passed to query="" from causing {{points}} to update.

I tried

`<ma-point-query ng-if="searchQuery==''" query="{name:searchQuery}" start="0" points="points" promise="promise"></ma-point-query>`

but it prevents the query from working at all. I'm considering editing the directive to not allow blank strings to return all points but it might be nice to provided a way to turn this functionality on/off with a boolean on the directive API

jazdw, 2016-05-16:
You've run into the child scope problem with `ng-if`, thats why it isn't working. Try this. I've also changed the `ng-if` to just `ng-if="searchQuery"` so that it works when the string is undefined. Another thing you may want to consider is adding the limit attribute.

````
<div ng-init="page={}"></div>
<input ng-model="searchQuery">
<ma-point-query ng-if="searchQuery" query="{name:searchQuery}" start="0" limit="10" points="page.points" promise="promise"></ma-point-query>
<div ng-repeat="pt in page.points track by pt.id">{{pt.deviceName}} - {{pt.name}}</div>
````

WillGeller, 2016-05-18:
Seems like child scoping on points was the issue. Got it to work with:

`<div ng-init="query={limit:15, page:1}; page={tableOrder:'-val', searchQuery:'', points:[], promise: {}}; showSearch=false; selected=[];"></div>
<ma-point-query ng-if="page.searchQuery" query="{name:page.searchQuery}" start="0" points="page.points" promise="page.promise"></ma-point-query>`

One other thing I am running into the need to have <ma-point-query> query on whether a point name contains 'power' & 'unit'.

How exactly do I go about setting up the query to work as such, it seems like the complexity of RQL is behind how the directive works.

closed 2016-05-18 by WillGeller

jazdw, 2016-05-18:
Sorry, i used syntax from version 3.1.x in my example. 3.0.1 doesn't have a `limit` attribute on `<ma-point-query>`. You have to specify it in the query attribute as shown below.

To do the query you described in 3.0.1 you would use the following:
````
<ma-point-query query="name=like=*power*&amp;name=like=*unit*&limit(15,0)" points="page.points"></ma-point-query>
````

In 3.1 you can do the same by adding single quotes around the query
````
<ma-point-query query="'name=like=*power*&amp;name=like=*unit*&limit(15,0)'" points="page.points"></ma-point-query>
````
Or you could use the new syntax, using an object to query.
````
<ma-point-query query="{name:'*power*unit*'}" limit="15" start="0" points="page.points"></ma-point-query>
````

WillGeller, 2016-05-18:
I am on 3.1.x (unofficial release) so the limit property does work although I don't think I will use it and leave it up to the query to limit the results. The new syntax, using an object to query didn't seem to work, maybe I'm not on the very latest. I used the second option:

`<ma-point-query ng-if="page.searchQuery" query="'name=like=*power*&amp;name=like=*unit*'" start="0" points="page.points" promise="page.promise"></ma-point-query>`

Actually since I need to get a bit dynamic with the searchQuery I ended up with something working like this:

`<ma-point-query ng-if="page.searchQuery" query="'name=like=*'+page.searchQuery+'*&amp;name=like=*unit*'" start="0" points="page.points" promise="page.promise"></ma-point-query>`

Although a bit verbose it is accomplishing what I need for the dashboard

jazdw, 2016-05-19:
The following should work. I'll email you the latest build so you can test.
````
<ma-point-query query="{name: page.searchQuery + '*' + page.unit}" limit="15" start="0" points="page.points"></ma-point-query>
````
But yeah you can't have two separate predicates for name when using the object query, you have to combine them into one and use a wildcard in the middle so order is important.

PhillipDWeeks, 2017-02-19:
Does anyone know why this is not working:? trying to return two points
<ma-point-query query="{$and: true, deviceName: dvName, name:'Buffer Tank Temp*Air Temp'}"  limit="2" points="points1b"></ma-point-query>
is there an or operator?