MangoAutomation/ma-dashboards #14: <ma-serial-chart> changing chart type adds Series-0

WillGeller, 2016-05-18:
I am using a dropdown to change the chart type on the graph, but it seems to add in a mysterious Series-0 to the legend

`<ma-serial-chart class="animate-hide" style="height: 540px; width: 100%" points="points" values="chartValues" default-type="{{chartType}}" default-axis="right" options="{legend:{}, chartCursor:{categoryBalloonDateFormat:'YYYY-MM-DD HH:NN'}}"> </ma-serial-chart>`

[image]

jazdw, 2016-05-18:
Is this with the 3.1 development branch or 3.0.1 release version? How are you retrieving the points?

jazdw, 2016-05-18:
Hmm, I can't replicate this in 3.1.x or 3.0.1. This is the code I used to test:
````
<div ng-init="page={}"></div>

<ma-date-range-picker from="page.from" to="page.to" preset="LAST_5_MINUTES" update-interval="1 minutes"></ma-date-range-picker>

<ma-point-query query="'name=like=*energy*&amp;name=like=*kwh*&limit(15,0)'" points="page.points"></ma-point-query>

<ma-point-values points="page.points" values="page.chartValues" from="page.from" to="page.to" rollup="AVERAGE" rollup-interval="1 minutes">
</ma-point-values>

<ma-serial-chart style="height: 540px; width: 100%" points="page.points" values="page.chartValues" default-type="step" default-axis="right" options="{legend:{}, chartCursor:{categoryBalloonDateFormat:'YYYY-MM-DD HH:NN'}}"> </ma-serial-chart>

````

WillGeller, 2016-05-18:
I am using 3.1.x development branch. I am using the point hierarchy to retrieve the points but the same happens if the points are passed over from a component using ma-query. I believe the bug happens when I update default-type="{{chartType}}".

Here are the 2 code blocks:

`<md-select ng-model="chartType" ng-init="chartType='smoothedLine'" flex="10">
        <md-option value="smoothedLine">
            Smoothed
        </md-option>
        <md-option value="line">
            Line
        </md-option>
        <md-option value="step">
            Step
        </md-option>
        <md-option value="column">
            Bar
        </md-option>
    </md-select>`

and 

`<ma-serial-chart style="height: 540px; width: 100%" points="points" values="chartValues" default-type="{{chartType}}" default-axis="right" options="{legend:{}, chartCursor:{categoryBalloonDateFormat:'YYYY-MM-DD HH:NN'}}">
                </ma-serial-chart>`

What is interesting is that the Series0 on the legend only comes up when I use the dropdown to update {{chartType}} but the bug doesn't occur when I update chartType on the scope via a $watch.

jazdw, 2016-05-19:
Well dont know what I was thinking yesterday, I wasn't even trying to update the default chart type in my example :confused: 

Should be fixed now, good to catch that before 3.1 is released.

closed 2016-05-19 by jazdw

WillGeller, 2016-05-20:
Great, thanks for the fix

commit ae31cead (2016-05-19): Fix bug where extra series shows up in graph, closes #14