MangoAutomation/ma-dashboards #30: Data Point Details - Time readout inconsistent

Puckfist, 2017-02-27:
I have observed the following:

If a load up a data point, sometimes I get a datetime and sometimes only a time in the "Time: " line of the page.

If I then navigate to a disabled point, the previous contents of the Time field remain.

If I am on the first data point, and it is displaying datetime, and I set a value, then it shall display time.

If I am getting time displayed, I can navigate to a disabled point, then turn the data source off and back on, then navigate back to the first point, then I will have a datetime displayed.

jazdw, 2017-03-29:
I've fixed the first part. The second part may be related but I couldn't really follow you exactly, can you please test again @Puckfist

jazdw, 2017-03-29:
Commit reference is 9a96f4956a8adfda39e2b614d49f481e3ac26989

jazdw, 2017-04-03:
I'm going to assume this is fixed.

closed 2017-04-03 by jazdw