MangoAutomation/ma-dashboards #33: project installation bug

nidhir-systango, 2017-03-23:
I have run the following commands(After cloning the project) 
1.)npm install(success)
2.)bower install(success)
3.)gulp

but the script automatically finished after few ms.
Starting 'default'...
Finished 'default' after 'x' ms

Any help would be great.
Thanks in advance.

jazdw, 2017-03-29:
@nidhir-systango Sounds like its working to me, mine normally takes about 1s.

What are you trying to do though? That wont actually install a module in mango. To do that you need to run "mvn package" and copy the zip file from maven-target into your Mango /web/modules directory then start Mango.

nidhir-systango, 2017-03-30:
@jazdw : just want to install the project and do some custom development.

jazdw, 2017-04-03:
@nidhir-systango 
As mentioned you should run "mvn install" to create a module zip and then copy it to the mango modules folder. Once you have done this you can run "gulp watch-web" to automatically copy any web files you change into the mango modules folder. Ensure you set your MA_HOME environment variable to the Mango installation root directory first. If you have any further questions please direct them to our forum. http://infiniteautomation.com/forum/

closed 2017-04-03 by jazdw