MangoAutomation/ma-dashboards #34: Edit menu - Template URL items bork page

Puckfist, 2017-03-28:
On a perfectly clean database (or presumably one with stuff), one can mess the menus up (either bad CSS (I had my font change) or no menu items render, or only icons render)

create a new menu item, place it at the top level. The state name and url can be anything, same with the datebar setting. Turn off "link to custom page" and supply a template url (not valid, try "asdf"). Supply a menu item name. Save menu item, save menu. Refresh page, click the new menu item, voila

jazdw, 2017-03-29:
This occurs because all not-found URLs below /dashboards are rewritten to point to the mdAdmin/index.html file. This is required for the client side URL router to work.

Not much we can really do about it, it's advanced functionality, if the user supplies a correct url it works. Also if the referenced (incorrect) template isn't below mdAdmin (e.g. try /asdf) it displays an error.

jazdw, 2017-04-03:
Closing as wont fix.

closed 2017-04-03 by jazdw