MangoAutomation/ma-dashboards #38: Dashboard Designer - User Permissions/ Save and Update Feedback

terrypacker, 2017-03-29:
When signed in as a user with only the 'user' privilege I tried to create a page.  I was able to save the page once as there is something in the JSON data table for it, but every time I try to edit it I get a 401 response because I don't have the default permission 'edit-pages' that was set into the edit permission when I created it.

The other problem is that there is no feedback on the page that my save isn't working, it appears that it is but when I go to view the page it is empty since I originally saved it without any components on it.  The only way I was able to tell there was a problem was to look at the 'Network' tab in dev tools.

jazdw, 2017-03-29:
I moved the second part into a separate issue.

Yep, there is an issue with the JSON store that a user can not edit a page that they created. If you do not set a permission then only the admin can edit it, if you set the permission to `users` then everyone can edit it.

Is there a permission which controls if a user can save to the JSON store in the first place?

terrypacker, 2017-03-29:
There is no special permission for saving, anyone can.  We should probably add some code that ensures that the user has edit permission to the data if they are creating a new one before they can save it?

jazdw, 2017-03-30:
I would say the way to fix this is to add a owner id column, a user should always be able to edit something they created.

terrypacker, 2017-03-30:
I agree with this and will add that.  It shouldn't affect the front end since the owner will be set in the REST calls.

terrypacker, 2017-03-30:
I'm adding a userId column to the table, making it not null and a Foreign key to users(id).  I am not however going to make it ON DELETE CASCADE so the entry will remain even if the user is deleted.

terrypacker, 2017-03-30:
The upgrade will be problematic however since we must assign a userId to all existing entries.  At this point I'm happy to set them all to 1 (admin).  Any objections to this?

terrypacker, 2017-03-30:
Change of plans here.  Added validation to the endpoints so that the User saving the data MUST have editPermission, if not then validation errors are returned *422* Status Code.

closed 2017-04-03 by jazdw