MangoAutomation/ma-dashboards #39: Data Point Details - Extra events displayed

Puckfist, 2017-03-29:
For the very first point (id: 1) I created I had the user's login events showing up in their events table for my admin's login events. Is it possible there is a missing &typename='DATA_POINT' or some such?

jazdw, 2017-03-29:
@Puckfist can you check the network tab and get me some more details.

Puckfist, 2017-03-29:
I see two requests to events,

http://localhost:8080/rest/v1/events?dataPointId=1&limit(5,0)  <-- returns some system events
http://localhost:8080/rest/v1/events?dataPointId=1&acknowledged=false&limit(0)  <-- limit 0?

Assuredly an API problem

terrypacker, 2017-03-29:
I'll take a look at the REST code and let you know what I find.

WillGeller, 2017-03-29:
The 2 requests is not an API problem, the first gets the events to display on the table (limited count) the second get's the total count of non-acknowledged events matching the other filters, the limit 0 prevents actual events from returning but I still use the .$count to display on the acknowledgeAll button

terrypacker, 2017-03-29:
I am working on a fix for this, there is a problem with the backend where by supplying the parameter 'dataPointId' it doesn't actually enforce that the results will be Data Point events.  To make a long story short 'dataPointId' is a shortcut that maps the query parameter to the column typeRef1.  Since events can be of many different types typeRef1 may not always refer to a data point ID, it could be a data source ID or some other ID from a different table.  

The fix I will implement will ensure that you get data point events for the above queries so there won't be any changes required in the UI when I'm done.

terrypacker, 2017-03-30:
So I modified the code to force the typeName='DATA_POINT' if one is querying by using the dataPointId parameter, which seems ok.  However there is another problem here:

There is a query parameter mapping for dataSourceId which we will have to remove because DATA_POINT type events have the dataSourceId as typeRef2 and DATA_SOURCE events have the dataSourceId as typeRef1.  

So for example, any existing queries should be changed so that:

To get data source events for data source with id 7 use this:
/rest/v1/events?eventType=DATA_SOURCE&typeRef1=7

To get data point events for data source with id 7 use this:
/rest/v1/events?eventType=DATA_POINT&typeRef2=7

We will need to discuss this and ensure the front end doesn't use the dataSourceId mapping since I am removing it.

jazdw, 2017-03-30:
OK, so our requests from the data point page as Phillip posted should work fine now?

Thanks Terry.

terrypacker, 2017-03-30:
Yes.  You can close this issue after testing my fix which seems to work fine in my light testing.  

I also looked around for anywhere we would be using Data Source events and didn't find anything so I think that bug is now fixed but wouldn't have been exposed until we redo the data source page.

closed 2017-04-03 by jazdw