MangoAutomation/ma-dashboards #42: Dashboard Designer - Save and Update Feedback

jazdw, 2017-03-29:
Originally raised by @terrypacker in #38 
> The other problem is that there is no feedback on the page that my save isn't working, it appears that it is but when I go to view the page it is empty since I originally saved it without any components on it. The only way I was able to tell there was a problem was to look at the 'Network' tab in dev tools.

WillGeller, 2017-07-21:
I added a saved toast to these pages as feedback when save is successful, via 07b4c857eb98e474623d01c8496ea9dab681aaf6, can this issue be closed or is there some other error saving logic needed?

jazdw, 2017-09-11:
There should be a error toast created in the error callback of the save promise.

WillGeller, 2017-09-11:
updated with error toast via 485afac05d2116cff672e4cdcc80631f1987a1c2

closed 2017-09-11 by WillGeller

WillGeller, 2017-09-15:
Feedback on playarea would be nice as well

jazdw, 2017-09-18:
I've improved the UI settings page on Friday, I'll add the toast messages in today.