MangoAutomation/ma-dashboards #44: Edit menu - 'dashboard' state item bad behavior

Puckfist, 2017-03-29:
There is a menu item defined by default with a state name of 'dashboard' and no label text. If one enables this and places any items nested within it, one will see odd menu behavior while navigating around. For instance, almost all links in the menu will cause this menu to expand. Closing another expanded menu item will close this item (if open).

jazdw, 2017-03-29:
Yeah you can definitely fuck things up by moving menu items to places where they shouldn't be, we really need to simplify the menu editor so you can't move the default items around as this simply doesn't work.

jazdw, 2017-04-24:
Menu editor has been reworked, even if you enable the dashboards/ui state now it doesn't show up in the menu so this can be closed. We have also locked down the state names so you can only create a state inside the selected parent item.

closed 2017-04-24 by jazdw