MangoAutomation/ma-dashboards #48: Adding Comments to Events - Where did it go?

terrypacker, 2017-03-30:
I can add a comment to an event from 2 locations, the events list on the data point details page and from the events page/table.  Both places accept my comment after I click on the flag and add some text but I have no idea where to view it?

terrypacker, 2017-03-30:
I can confirm that the comment does get added to the DB by hitting the user comment end point at:

rest/v1/comments

WillGeller, 2017-03-30:
> @WillGeller <https://github.com/WillGeller>.

What is the plan for how we want to display these event type comments? From
what I can remember we may have chosen to not display them and perhaps turn
off the clicking flag to add event comment

terrypacker, 2017-03-30:
On the legacy page the comments are displayed within the row of the event in the table. On the new page I would like to see some icon/indication in the row that there are comments available to view and then the User must click something to either:

1.  Expand row to show comments
2.  Show popup/modal with comments list in  it.

However this is probably not a huge priority for the 3.0.2 Core release.  I would either fix it up as above or do nothing.  I wouldn't bother removing the functionality of adding comments.

WillGeller, 2017-03-31:
When is the ETA for 3.0.2 release?

WillGeller, 2017-05-03:
Implemented as 'Expand row to show comments' via 8f95c02e8e33ee2f129329df1b7b6b4e21b96fee

closed 2017-05-03 by WillGeller