MangoAutomation/ma-dashboards #50: Missing help pages

jazdw, 2017-03-30:
Need some help pages for the following (will update as I find them)
* dashboard.help.users

WillGeller, 2017-05-02:
Other pages that need help pages:

- [ ] Add Device
- [ ] Dashboard Designer
- [ ] Config Import/Export
- [ ] System Settings?

WillGeller, 2017-05-12:
I added/updated help docs for the checked items in my previous comment. I probably won't get around to creating new ones for the last couple until I return to work but the ones that are new/updated could use a review and editing by another person ( @Puckfist ?)

WillGeller, 2017-09-07:
I am starting to go back in today and update help pages, starting with the new toolbar UI on the page editor, file store etc...

jazdw, 2020-09-01:
Outdated, new help system will be implemented.

closed 2020-09-01 by jazdw