MangoAutomation/ma-dashboards #53: User - Users can edit their own permissions!

Puckfist, 2017-03-30:
Currently a user can navigate to the users page (it may require typing in the URL) and modify their own permissions! Only superadmins should be able control privileges. Terry indicated he intended to put more protections in the API for this as well.

terrypacker, 2017-03-30:
Fixed in the Api here:  infiniteautomation/ma-modules-public@b7b0520de096a84a6f140a85c5a186d411827eff

jazdw, 2017-03-31:
Holy crap, how long has that been allowed through the API?
Terry re. your fix, its possible that the two permission strings aren't equal but are equivalent. Just another fun side effect of using a comma separated string for our permissions. Do you think it might be better to just ignore the new permission string if the user is not admin?

7ee1020cea6649e22c1192c85bbd3f57499fda23 disables the input if not admin.

terrypacker, 2017-03-31:
I used 'equals' because I figure they shouldn't be able to change even the order of the permissions.  I'm going to leave it as is because I think ignoring it might be misleading if you get back a 200 and expected your permissions to change.  The only other thing I can think of is to still do the validation on the new permissions but just never set them in the user that gets saved, however that seems to be the same end result to me.

closed 2017-04-03 by jazdw