MangoAutomation/ma-dashboards #56: Watchlist chart data export - time column not right

Puckfist, 2017-03-31:
Exporting data from the chart on the new watch list, I get rows like:

"2017-03-31","5.292705174164186","5.293","0","0"

Where the time column only has date information. If the user instead uses the download button they'll get data like (multiple row):

"AVERAGE","1490994230000","5.292705174164186",

Where exports from the legacy watchlist or data point details gives times as

2017/03/31 14:56:50.557

Which I believe is the time expected by the data import module

Puckfist, 2017-03-31:
It looks like JSON from the download button uses epoch times outside of quotes (so, as numbers) while the JSON from the chart gives "2017-03-31T21:04:30.000Z" style stamps

jazdw, 2017-04-02:
> Exporting data as csv from the chart on the new watch list, I get rows like:
The format of this data is determined by amCharts and yes its a bit borked. It will definitely not give data in the same format as the download button. We should probably just hide the option.

> Which I believe is the format expected by the data import module
The Point Data Import module uses DateTimeFormat.forPattern("yyyy/MM/dd HH:mm:ss.SSS").parseDateTime()

The data import seems to be very picky, it requires all 8 headers when "xid","value","timestamp" should suffice. It also should support epoch timestamps and ISO formatted dates IMO. Really we should just replace this module with a simple new UI page and add a point value import end-point to the API module.

jazdw, 2017-04-03:
I've changed the date format in the amcharrts export to use an ISO formatted date. It's still not compatible with the import but I believe these export options might still be useful to people so I've left the option there. 84c7cc6e00c035480441fa1634e42466ebb5e516

In regards to the Watch-list data download format, this is an API change.

closed 2017-04-03 by jazdw