MangoAutomation/ma-dashboards #60: No feedback on Configuration Import

Joelhaggar, 2017-04-04:
I imported a json configuration and saw the bar across the top complete but never got any out up of what was imported or confirmation that it was complete.

jazdw, 2017-04-05:
Can't replicate this, I always get the messages when the import completes. Unless it was an import of an empty array or something.

There is an issue with the API in that no messages are returned while the import is in progress, i.e. I might be 50% of the way through importing 1000 data points but I get no messages, only a status of 50%. Only when it reaches 100% are the messages returned.

jazdw, 2017-04-05:
Opened issue infiniteautomation/ma-core-public#978

Joelhaggar, 2017-04-05:
This issues seems to be that if there is an error on one of the items imported no results will be displayed.  I tested this by importing a Meta data point that I know would fail.

terrypacker, 2017-04-06:
This should now work with the fix from  infiniteautomation/ma-core-public#978

closed 2017-04-07 by jazdw