MangoAutomation/ma-dashboards #63: DashboardDesigner - "Advanced CSS" can fight other property tabs

Puckfist, 2017-04-11:
It is possible to have a conflict between the advanced CSS properties defined and the other tabs for setting properties. The one I found was the background color. If I added the CSS property 'background-color' then give it the value 'purple' I see my square turn purple in the viewer. Then I click back on the background tab in the property editor and POOF my purple background has disappeared!

Puckfist, 2017-04-11:
I found another fight in the border property. If I set the border radius through the property tab, I see border radii for each corner added to the Advanced CSS. If I modify them there, now the border tab's border radius is empty. Now I edit that again, it becomes the new border radius applied, I go back to the advanced CSS tab and the corner radii properties still exist!

Puckfist, 2017-04-11:
I was having trouble further reproducing the background color issue

jazdw, 2017-04-11:
I also cannot reproduce the background color issue. As for the border radius, yes if all four border radius are not set to the same value the shorthand CSS property "border-radius" will not be set.

closed 2017-04-11 by jazdw