MangoAutomation/ma-dashboards #74: UI Bug in Safari

WillGeller, 2017-05-31:
new UI in safari and the home page doesn’t seem to be responsive?  It is fine in Chrome.
[image]

WillGeller, 2017-05-31:
Seems to be a general issue with safari and the flex layout wrap stuff we use on many pages:
[image]
[image]
[image]
[image]
[image]
[image]

WillGeller, 2017-05-31:
https://github.com/angular/material/issues/10516

Same issue even on official angular material docs

jazdw, 2017-05-31:
This is a Safari bug, they haven't pulled in a fix from WebKit from the sounds of it. I dont think its worth our time trying to work around it.

WillGeller, 2017-05-31:
I agree that it would be difficult to work around but we should be clear that basically the app runs best in chrome and may not work perfectly in other evergreen browsers.

Another issue with ignoring is it may cause problems with the mobile view on iPhones/iPads

terrypacker, 2017-05-31:
I agree with Will its important for us to support iPhone/iPads.  Maybe dig a little deeper to see how big a deal this is or how soon they may pull in a fix?  I just updated my iPad yesterday and it doesn't work with the UI any better than Safari on OSX.

jazdw, 2017-06-15:
@Joelhaggar and I just talked about this, we tried using the Chrome browser on iOS but no cigar, it uses Apple's rendering engine and is just as broken.

WillGeller, 2017-06-16:
Many of our clients have to use the app on iPads/iPhones as a minimal requirement. I suggest we use the CSS override file mentioned above. It may be possible to detect if running on iOS or safari and load the CSS file conditionally to avoid downloading it on devices running chrome. Minifying the above file gets it down to 87k

WillGeller, 2017-07-18:
Testing in Safari 10.0.3 Desktop it seems like the overrides CSS file is loading and is fixing a lot of the rendering issues I still see a few though.

Edit Menu Dialog doesn't display content:
[image]

Users Page on Mobile, top dropdown cut off:
[image]

WillGeller, 2017-07-20:
Added a commit that fixed edit dialog rendering in Safari 6ddc2243648908c53369767ec7faa8fde2b3bd4f

commit 394556ca (2017-07-17): Add a fix for safari flex box rendering

Closes #74
commit 8afcdb67 (2017-07-18): Add a fix for safari flex box rendering

Closes #74
closed 2017-07-20 by WillGeller