MangoAutomation/ma-dashboards #75: Review of Examples and Component Docs

WillGeller, 2017-05-31:
- check Examples and update where needed, add new examples showing usage of newer components and techniques.
- Dashboard designer examples?
- Possibly include videos
- Update directive documentation and add new components available to users
- Consider moving old template pages under examples to support site and out of app to avoid confusion

WillGeller, 2017-06-06:
I went through and reviewed/updated all existing directive docs and examples.

A couple where I was unsure on some attribute usage that need updating. Can you take a look at these, @jazdw :

- ma-filtering-point-list
- ma-fn
- ma-point-query

In addition there are a bunch of new directives/components/services that haven't had any docs written for them yet. I think we should focus on getting docs written for just the HTML components that users will use in pages/dashboards. Services and app interface components might not be worth documenting in the public facing api-docs as they are mostly used internally.

jazdw, 2017-06-08:
Yeah there definitely are some pretty obscure directives that are more for internal use and are lower priority for documentation. I'll try and look at those ones you mentioned above this week.

closed 2017-06-26 by WillGeller