MangoAutomation/ma-dashboards #78: Error on import of demoMangoConfig.json

WillGeller, 2017-06-02:
Importing this file:
http://localhost:8080/modules/mangoUI/web/demoMangoConfig.json

errors on importing some of the data points and fails to import point hierarchy:

> Data source 'DS_997094': added
Data source 'vmeters': added
Data point 'DP_253536': added
Data point 'temperature': added
Data point 'DP_119588': added
Data point 'DP_984239': added
Data point 'DP_698831': added
Data point 'DP_278103': added
Data point 'DP_686983': added
Data point 'DP_549856': added
Data point 'DP_331467': added
Data point 'DP_806477': added
Data point 'DP_596850': added
Data point 'DP_370045': added
Data point 'DP_348223': added
Data point 'DP_287036': added
Data point 'DP_394777': added
Data point 'DP_347418': added
Data point 'DP_924123': added
Data point 'DP_222324': added
Data point 'DP_077116': added
Data point 'DP_390622': added
Data point 'DP_992791': added
Data point 'DP_268535': added
Data point 'DP_033512': added
Data point 'DP_840606': added
Data point 'DP_916776': added
Data point 'DP_440955': added
Data point 'DP_100375': added
Data point 'DP_279767': added
Data point 'DP_077192': added
Data point 'voltage': added
Data point 'multistate': added
Data point 'binary': added
Data point 'DP_355369': Error parsing 'integralUnit'
Data point 'DP_614346': Error parsing 'unit'
Data point 'DP_368591': Error parsing 'integralUnit'
Data point 'DP_241169': Error parsing 'integralUnit'
Data point 'DP_556556': Error parsing 'integralUnit'
Data point 'DP_622129': Error parsing 'unit'
Data point 'DP_364171': Error parsing 'integralUnit'
Data point 'DP_422050': Error parsing 'integralUnit'
Data point 'DP_805604': Error parsing 'integralUnit'
Data point 'DP_399285': Error parsing 'unit'
Data point 'DP_029774': Error parsing 'integralUnit'
Data point 'DP_202333': Error parsing 'integralUnit'
Point hierarchy: 'JsonException writing property 'subfolders' of class java.util.List', caused by: 'JsonException writing property 'subfolders' of class java.util.List', caused by: 'Data point with XID 'DP_614346' not found'{1}

jazdw, 2017-06-05:
@WillGeller Can you post the output from your log file. I'd say its because you didn't open the .json file as UTF-8

WillGeller, 2017-06-05:
> INFO  2017-06-05T11:30:10,568 (org.springframework.web.servlet.handler.AbstractUrlHandlerMapping.registerHandler:362) - Mapped URL path [/sqlConsole.shtm] onto handler of type [class com.serotonin.m2m2.web.mvc.UrlHandlerController] 
INFO  2017-06-05T11:30:10,568 (org.springframework.web.servlet.handler.AbstractUrlHandlerMapping.registerHandler:362) - Mapped URL path [/globalScripts.shtm] onto handler of type [class com.serotonin.m2m2.web.mvc.UrlHandlerController] 
INFO  2017-06-05T11:30:12,086 (com.infiniteautomation.system.MulticastDNSServiceDiscovery.startJmDNS:171) - Multicast DNS Service Discovery started on en0 (en0) with IP address 192.168.0.149 
INFO  2017-06-05T11:30:30,968 (com.serotonin.m2m2.rt.RuntimeManager.stopDataSourceShutdown:418) - Data source 'Dashboard Demo' stopped 
INFO  2017-06-05T11:30:30,981 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:376) - Data source 'Dashboard Demo' initialized 
INFO  2017-06-05T11:30:30,982 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:381) - Data source 'Dashboard Demo' took 4.930074ms to start 
INFO  2017-06-05T11:30:30,986 (com.serotonin.m2m2.rt.RuntimeManager.stopDataSourceShutdown:418) - Data source 'Virtual Meters' stopped 
INFO  2017-06-05T11:30:31,011 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:376) - Data source 'Virtual Meters' initialized 
INFO  2017-06-05T11:30:31,011 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:381) - Data source 'Virtual Meters' took 22.499749ms to start 
INFO  2017-06-05T11:32:46,015 (com.serotonin.m2m2.rt.RuntimeManager.stopDataSourceShutdown:418) - Data source 'Dashboard Demo' stopped 
INFO  2017-06-05T11:32:46,023 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:376) - Data source 'Dashboard Demo' initialized 
INFO  2017-06-05T11:32:46,023 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:381) - Data source 'Dashboard Demo' took 4.674058ms to start 
INFO  2017-06-05T11:32:46,025 (com.serotonin.m2m2.rt.RuntimeManager.stopDataSourceShutdown:418) - Data source 'Virtual Meters' stopped 
INFO  2017-06-05T11:32:46,039 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:376) - Data source 'Virtual Meters' initialized 
INFO  2017-06-05T11:32:46,040 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:381) - Data source 'Virtual Meters' took 13.330395ms to start 
INFO  2017-06-05T11:34:00,796 (com.serotonin.m2m2.rt.RuntimeManager.stopDataSourceShutdown:418) - Data source 'Dashboard Demo' stopped 
INFO  2017-06-05T11:34:00,801 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:376) - Data source 'Dashboard Demo' initialized 
INFO  2017-06-05T11:34:00,801 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:381) - Data source 'Dashboard Demo' took 3.172642ms to start 
INFO  2017-06-05T11:34:00,802 (com.serotonin.m2m2.rt.RuntimeManager.stopDataSourceShutdown:418) - Data source 'Virtual Meters' stopped 
INFO  2017-06-05T11:34:00,812 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:376) - Data source 'Virtual Meters' initialized 
INFO  2017-06-05T11:34:00,812 (com.serotonin.m2m2.rt.RuntimeManager.initializeDataSourceStartup:381) - Data source 'Virtual Meters' took 9.348334ms to start 

I just tried importing the the json file using the 'Upload JSON File' option where you select the file and it worked. It fails when I copy and paste in the json to upload

closed 2017-07-17 by jazdw