MangoAutomation/ma-dashboards #92: <ma-point-statistics> errors when passing in points array initially empty

WillGeller, 2017-09-22:
```
TypeError: Cannot read property 'length' of undefined
    at $scope.$watch.changedXids.added (pointStatistics.js?v=1504670134:120)
    at m.$digest (
```

When using the following code:

```
<ma-watch-list-get watch-list-xid="{{$ctrl.watchlistXid}}" ng-model="$ctrl.watchList" on-points-change="$ctrl.dataPoints = $points"></ma-watch-list-get>
<ma-point-statistics  points="$ctrl.dataPoints" from="$ctrl.dateBar.data.from" to="$ctrl.dateBar.data.to" statistics="$ctrl.statsArray" rendered="false"></ma-point-statistics>
```

Resolved by adding ng-if
```
<ma-watch-list-get watch-list-xid="{{$ctrl.watchlistXid}}" ng-model="$ctrl.watchList" on-points-change="$ctrl.dataPoints = $points"></ma-watch-list-get>
<ma-point-statistics  ng-if="$ctrl.dataPoints.length" points="$ctrl.dataPoints" from="$ctrl.dateBar.data.from" to="$ctrl.dateBar.data.to" statistics="$ctrl.statsArray" rendered="false"></ma-point-statistics>
```

jazdw, 2017-09-25:
@WillGeller I can't replicate this or find a reference to where we are checking the length of changedXids.added. Are you using the latest UI module? Maybe call me.

WillGeller, 2017-09-25:
I noticed it on Duke Energy and looking now, they are not on the latest MangoUI module.  Looks like they are on 3.2.1 and I put in a fix for the issue on 7/14/17 ad94192b7c7ab984e5ca0253ab6154360bd54119
I am assuming this will be resolved when we update them

closed 2017-09-25 by WillGeller

jazdw, 2017-09-25:
OK cool. Thanks.