MangoAutomation/ma-dashboards #111: SQL overload due to Watchlists getting active events for all data points every 5 seconds

jazdw, 2017-11-02:
@Joelhaggar the changes Will made to the watchlists to get the active events for all the data points is causing issues on watchlists with a lot of points.

@Puckfist discovered this on a client system, there were hundreds of REST requests occurring and hitting the SQL data base all at once.

We should be able to get a list of active events for a datapoint from the EventManager cache I believe. Is this true @terrypacker ?

jazdw, 2017-11-02:
This request occurs every 5 seconds for every data point - `/rest/v1/events?dataPointId=59&active=true&limit(0,0)`

jazdw, 2017-11-02:
The reason this request takes so long is that there is not an index on the typeRef1 column in the events table.

terrypacker, 2017-11-02:
Active events are all cached in the UserEventCache within the EventManager.  However we have not yet implemented querying the cache (but it wouldn't be very hard with the RQLToObjectListQuery I don't think.)

The other way to achieve this might be to perform the query once when the page loads and then register for the events over a web socket and update the page that way.

jazdw, 2017-11-02:
I agree we should use the websocket, that's something I can change right now.

terrypacker, 2017-11-02:
I've added a new endpoint for v2 that uses the UserEventCache, it can only return events that are un-silenced for a user.  See EventDao.getAllUnsilenced for the data the cache is built with.

The following query will return every data point event:

`rest/v2/user-events?eventType.eventType=DATA_POINT`

The following query will return every data point event for a data point with ID:
`rest/v2/user-events?eventType.eventType=DATA_POINT&eventType.dataPointId=1099`

The following query will return every data point event for data points in the set:

`https://localhost:8443/rest/v2/user-events?eventType.eventType=DATA_POINT&in(eventType.dataPointId,1099,1100)`

jazdw, 2017-11-07:
@terrypacker I haven't used that new end-point. I didn't want to deal with a new URL, parameters and model.

I've made a fix in 8eb6b6d24fdd676908b630334d1e7c393b5b00b3 which just queries for events when the directive is initialized and then listens on the websocket for events being raised and returned to normal.

jazdw, 2017-11-07:
@terrypacker also what does a websocket event type 'DEACTIVATED' mean? Also is it correct to assume that event for which "return is not applicable" will never have "active: true" in the model?

terrypacker, 2017-11-07:
DEACTIVATED events happen when a data point, data  source, or publisher is either deleted or disabled.  We should document these in the code much better as I had to do a fair bit of tracing to figure that out.  I'll take care of that in the core code.

Yes any event that is not rtn applicable will never be active:
`active = rtnApplicable && rtnTimestamp == 0`

jazdw, 2017-11-09:
OK thanks. I think I noticed a Websocket RETURN_TO_NORMAL message yesterday that had `active: true`.. Might look into that.

jazdw, 2017-12-05:
@dafrazier

jazdw, 2017-12-12:
Closing this as the main issue is fixed. Index as per issue linked above should help the initial query speed.

closed 2017-12-12 by jazdw