MangoAutomation/ma-dashboards #159: System Settings Validation Failed Messages Not Shown.

terrypacker, 2018-08-30:
I've fixed up the Emport V2 Rest Controller to properly return validation messages when things go wrong.  The problem was that there is no 'property' in the message i.e. the context-key that would normally be used to show what input was invalid.  So the new format looks like this when you try to import a data source with an invalid value:

```
{
  "resourceId" : "IMPORT_f3001222-190f-4fd7-a222-c54fd457c03e",
  "start" : "2018-08-30T11:23:50.980-10:00",
  "finish" : "2018-08-30T11:23:51.263-10:00",
  "state" : "COMPLETED",
  "progress" : 100.0,
  "owner" : "admin",
  "validationMessages" : [ {
    "level" : "ERROR",
    "message" : "Data source 'DS_aa23d4f2-b876-42b6-b0a9-7ce982c82d2b': Invalid 'purgeType' value of 'ANNOS'. Valid values are [MILLISECONDS, SECONDS, MINUTES, HOURS, DAYS, WEEKS, MONTHS, YEARS]",
    "property" : null
  } ],
  "genericMessages" : [ "Data source 'DS_997094': saved", "Data source 'DS_c074fb28-89ef-45a4-a5a8-43133e7ffb45': saved", "Data source 'internal_mango_monitoring_ds': saved", "Data source 'DS_9e99f60b-9b89-4470-a5f2-af330a6c0cf0': saved", "Data source 'DS_df1991b2-e103-406b-b5b1-23669c70526d': saved", "Data source 'DS_2d9d3960-f48f-4a39-8096-a3d6eb051b96': saved", "Data source 'DS_c1c58810-31f7-4d09-9eea-923e39bb27d7': saved", "Data source 'vmeters': saved" ],
  "threadName" : "Temporary Resource Timeout for : IMPORT_f3001222-190f-4fd7-a222-c54fd457c03e",
  "taskId" : "TR_IMPORT_f3001222-190f-4fd7-a222-c54fd457c03e",
  "expires" : 1535664531263,
  "queueSize" : 0
}

terrypacker, 2018-08-30:
Note the validationMessages has a null property field, which is how the Importer works so without changing the core this is how it will work for now.

jazdw, 2018-08-31:
Duplicate of #126 as per Asana task.

closed 2018-08-31 by jazdw