MangoAutomation/ma-dashboards #170: Sent test email - better feedback

terrypacker, 2018-12-12:
I just modified the /v2/server/email/test endpoint to provide immediate feedback from the sending email if it fails.  Currently the page only shows 'Internal Server Error' but the endpoint is now throwing its exceptions.

terrypacker, 2018-12-12:
We can discuss this but as of now if you try to test an email on a port that doesn't exists the endpoint's call will not return until the socket timeout is hit.  Sounds like a use for temporary resources.

terrypacker, 2018-12-12:
I have added a way to get the session debug information for the email session so we can add that to the model.  It is currently just logged in the controller.  I figure we can throw a new type of exception with the debug output in it or something.

jazdw, 2019-01-03:
@terrypacker OK so what do you need from me? I'm assuming the front end is displaying the messages from the new exceptions already?

terrypacker, 2019-01-03:
@jazdw, you could in theory provide more information.  The message on the bottom toaster/popup is 

```
Error sending email to 'admin@yourMangoDomain.com' — Internal Server Error
```

But the JSON response is:
```
{
  "cause" : "MailSendException: Mail server connection failed; nested exception is com.sun.mail.util.MailConnectException: Couldn't connect to host, port: localhost, 25; timeout 60000;\n  nested exception is:\n\tjava.net.ConnectException: Connection refused (Connection refused). Failed messages: com.sun.mail.util.MailConnectException: Couldn't connect to host, port: localhost, 25; timeout 60000;\n  nested exception is:\n\tjava.net.ConnectException: Connection refused (Connection refused)",
  "mangoStatusCode" : -1,
  "mangoStatusName" : null,
  "localizedMessage" : "Internal Server Error"
}
```

terrypacker, 2019-01-03:
I could include the STMP session debug information if we wanted.  For example:

terrypacker, 2019-01-03:
Or we could go a step further and provide the session output in the response every time.

jazdw, 2019-01-03:
I would agree that we should provide more information, I thought the localizedMessage would be customized in the exception response. I would create a new REST exception with the full error information in its model. I can then display this on the page.

I do not use the cause field anywhere in the UI, its something I think should be removed. It leaks information about the system. Or at least make it only visible for admins.

terrypacker, 2019-04-09:
Bump.  This could make it into the UI for 3.6 maybe?

[https://forum.infiniteautomation.com/topic/4143/fail-to-receive-email-from-mango/2](https://forum.infiniteautomation.com/topic/4143/fail-to-receive-email-from-mango/2)

jazdw, 2019-04-11:
Few things..
* `new TranslatableMessage("common.default"..)` is an evil hack and should not be a thing
* Are you expecting me to use localizedMessage or smtpSessionLog? Both are hideously long and unwieldy. Not sure why we have two things that are essentially the same (see below).
* My connect timeout is set to 60s, however I received the response below back from the REST API after 21s so it seems the timeout is not being set correctly

```
cause: null
localizedMessage: "Mail server connection failed; nested exception is com.sun.mail.util.MailConnectException: Couldn't connect to host, port: smtp.gmail.com, 588; timeout 60000;↵  nested exception is:↵	java.net.ConnectException: Connection timed out: connect. Failed messages: com.sun.mail.util.MailConnectException: Couldn't connect to host, port: smtp.gmail.com, 588; timeout 60000;↵  nested exception is:↵	java.net.ConnectException: Connection timed out: connect"
mangoStatusCode: 5000
mangoStatusName: "GENERIC_500"
smtpSessionLog: "DEBUG: getProvider() returning javax.mail.Provider[TRANSPORT,smtp,com.sun.mail.smtp.SMTPTransport,Oracle]
↵DEBUG SMTP: useEhlo true, useAuth true
↵DEBUG SMTP: trying to connect to host "smtp.gmail.com", port 588, isSSL false
↵"
```

terrypacker, 2019-04-12:
- I have fixed the common.default hack, was more of a placeholder for this issue.
- I would expect you to display the localized message in the toaster on the bottom of the page
- I would expect that if an email fails to send you provide the option to expand a box with the SMTP session log output
- The timeout appears to be working for me (Set it to 10s as the UI has an inbuilt timeout < 60s it seems).

I did however see that the UI cancels the requests if it doesn't get a response back in some amount of time that is much less than 60s (maybe around 21s). The message is this:
`Error sending email to 'admin@yourMangoDomain.com' — Request timed out`

terrypacker, 2019-04-12:
Localized message cleanup commit: https://github.com/infiniteautomation/ma-modules-public/commit/6e609cf1bfd21ff1fc021f4d7c8092882afec47a

jazdw, 2019-04-13:
@terrypacker The UI request timeout is 30s so I'm pretty sure that is not it. But I'll double check.

terrypacker, 2020-02-27:
The test email endpoint returns this on failure, please display the smtpSessionLog if it exists in the response.  There are situations where the log won't exist like if the from email address is invalid.

```
{
  "cause" : "EmailFailedException: Email send failure",
  "smtpSessionLog" : "DEBUG: getProvider() returning javax.mail.Provider[TRANSPORT,smtp,com.sun.mail.smtp.SMTPTransport,Oracle]\nDEBUG SMTP: useEhlo true, useAuth false\nDEBUG SMTP: trying to connect to host \"smtp.gmail.com\", port 25, isSSL false\n220 smtp.gmail.com ESMTP 199sm8450098pfu.71 - gsmtp\nDEBUG SMTP: connected to host \"smtp.gmail.com\", port: 25\nEHLO localhost\n250-smtp.gmail.com at your service, [204.210.127.189]\n250-SIZE 35882577\n250-8BITMIME\n250-STARTTLS\n250-ENHANCEDSTATUSCODES\n250-PIPELINING\n250-CHUNKING\n250 SMTPUTF8\nDEBUG SMTP: Found extension \"SIZE\", arg \"35882577\"\nDEBUG SMTP: Found extension \"8BITMIME\", arg \"\"\nDEBUG SMTP: Found extension \"STARTTLS\", arg \"\"\nDEBUG SMTP: Found extension \"ENHANCEDSTATUSCODES\", arg \"\"\nDEBUG SMTP: Found extension \"PIPELINING\", arg \"\"\nDEBUG SMTP: Found extension \"CHUNKING\", arg \"\"\nDEBUG SMTP: Found extension \"SMTPUTF8\", arg \"\"\nDEBUG SMTP: use8bit false\nMAIL FROM:<terry@infiniteautomation.com>\n530 5.7.0 Must issue a STARTTLS command first. 199sm8450098pfu.71 - gsmtp\nDEBUG SMTP: got response code 530, with response: 530 5.7.0 Must issue a STARTTLS command first. 199sm8450098pfu.71 - gsmtp\n\nRSET\n250 2.1.5 Flushed 199sm8450098pfu.71 - gsmtp\nDEBUG SMTP: MessagingException while sending, THROW: \ncom.sun.mail.smtp.SMTPSendFailedException: 530 5.7.0 Must issue a STARTTLS command first. 199sm8450098pfu.71 - gsmtp\n\n\tat com.sun.mail.smtp.SMTPTransport.issueSendCommand(SMTPTransport.java:2358)\n\tat com.sun.mail.smtp.SMTPTransport.mailFrom(SMTPTransport.java:1823)\n\tat com.sun.mail.smtp.SMTPTransport.sendMessage(SMTPTransport.java:1300)\n\tat org.springframework.mail.javamail.JavaMailSenderImpl.doSend(JavaMailSenderImpl.java:464)\n\tat org.springframework.mail.javamail.JavaMailSenderImpl.send(JavaMailSenderImpl.java:360)\n\tat org.springframework.mail.javamail.JavaMailSenderImpl.send(JavaMailSenderImpl.java:377)\n\tat org.springframework.mail.javamail.JavaMailSenderImpl.send(JavaMailSenderImpl.java:365)\n\tat com.serotonin.web.mail.EmailSender.send(EmailSender.java:179)\n\tat com.serotonin.web.mail.EmailSender.send(EmailSender.java:170)\n\tat com.serotonin.web.mail.EmailSender.send(EmailSender.java:122)\n\tat com.infiniteautomation.mango.spring.messaging.EmailMessageTransport$1.execute(EmailMessageTransport.java:115)\n\tat com.serotonin.m2m2.rt.maint.BackgroundProcessingImpl$WorkItemRunnable.run(BackgroundProcessingImpl.java:550)\n\tat java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)\n\tat java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)\n\tat java.base/java.lang.Thread.run(Thread.java:830)\nQUIT\n221 2.0.0 closing connection 199sm8450098pfu.71 - gsmtp\n",
  "mangoStatusCode" : 5000,
  "mangoStatusName" : "GENERIC_500",
  "localizedMessage" : "Email send failure"
}
```

jazdw, 2020-02-28:
The timeout setting does seem to work if you set it below 21 seconds, but anything higher than that and it times out after 21 seconds. No idea why.

jazdw, 2020-02-28:
Closed in 5752322a7423a3bee33ead50d1c1cbb4ea06dce2

closed 2020-02-28 by jazdw