MangoAutomation/ma-dashboards #183: Check X-Mango-Default-URI-Required Header on Login

terrypacker, 2019-02-27:
When the admin user logs in for the first time ever or when the license version changes the `X-Mango-Default-URI-Required=true` header is returned.  This indicates that the URI in the `X-Mango-Default-URI` header should take priority as the login destination.

See:

infiniteautomation/ma-core-public@22f687f0e789841ab0efdbebf454a1b1d00b3804

infiniteautomation/ma-modules-public@a8b78b28a93c499864606d532c40bbea02c50ff4

terrypacker, 2019-02-27:
Also note I added the `X-Mango-Last-Upgrade` and `X-Mango-Default-URI-Required=true` headers to the switch user and un-switch user response.

jazdw, 2019-03-08:
Closed via bdc3a1908f3b54a371637a0dab883e1a34995778

closed 2019-03-08 by jazdw