MangoAutomation/ma-dashboards #189: Excel Reports Validation Messages for named ranges not working

terrypacker, 2019-03-19:
[image]
 
As can be seen in the image if you use the same named range you can get back a validation message that should be tied to the form.  Something isn't working in the form or the format isn't quite right coming back from the REST endpoint.

terrypacker, 2019-04-09:
Further update here.  The validation response is displayed in a popup on the bottom of the page:

```
Report Template not saved - Validation failed, property ''time-series-0.points[1].namedRange'' — Range with name last_month_power already assigned
```

terrypacker, 2019-04-09:
It looks like the backend uses 2 different types of index notation for the validation.  I tried to fix this but found that even when I used what I think is the correct notation the input is not highlighted.

`time-series[0].points[1].namedRange`

jazdw, 2019-04-11:
Yep, looks like the inputs are not named correctly in the markup. Now the validation messages have been fixed up I'll take another look.

Did you use the new method I added? infiniteautomation/ma-core-public@42ba94efb2cfec59f678a2540998247b9b9e6763

Can you reference the commit where you changed the validation messages please.

terrypacker, 2019-04-11:
I did not change the validation messages.  I said that I _tried_ to change them and found that it didn't make any difference so I didn't commit the changes.  I will fix them up and test this now.

terrypacker, 2019-04-11:
Also I cannot use the validate method you created because these validation methods require additional parameters to ensure things like other Time Series do not use the same named ranges.  We could consider adding additional parameters to the validate() methods?   Maybe a map or object that can be passed through and used in cascading validations like this?

terrypacker, 2019-04-11:
The validation changes were done in this commit: infiniteautomation/ma-core-public@3767a8fc4a022f320ccd473d9bc75b913dc05465

jazdw, 2019-04-11:
> Also I cannot use the validate method you created because these validation methods require additional parameters to ensure things like other Time Series do not use the same named ranges. We could [...]

Its not a bad idea, maybe we should discuss this tomorrow.

jazdw, 2019-04-13:
I have been working on this (infiniteautomation/ma-modules-proprietary@92482508dbfb24878899875d8ca16b45b80179f4) but have found numerous other bugs that I am working on fixing.

terrypacker, 2019-12-12:
@guetteluis I just re-tested this and it needs work.  If I manually type in a duplicate named range then the validation message comes back as required I think this input needs to be tied to the model for both the drop down part AND the manual entry part.

Also when I get the message back about the input being required I get a real ugly popup at the bottom of the page that says:

`Report Template not saved - Validation failed, property ''timeSeries[0].points[1].namedRange'' — Cannot have empty named range`

That should not happen so can you look into that I assume it has something to do with the `required` message already being applied.

closed 2020-02-18 by terrypacker