MangoAutomation/ma-dashboards #194: Event handler names were not always required - event detector popup

Puckfist, 2019-04-01:
If one has an event handler from the past, its 'alias' field was not required. Now it is required that an EH have a name, but old handlers appear in the event detector list (if unaliased originally) as blank space:

[image]

Perhaps it should say the handler type as the old page did? Maybe that with the XID? Or, maybe something like that should autogenerate the name field during the upgrade?

jazdw, 2019-04-02:
Sounds like a job for an upgrade script, I assume the DB column is not marked as "non-null" either.

terrypacker, 2019-04-08:
This is done via upgrade script and the new name is type description - XID.  I also marked the alias column in the DB to be not null

closed 2019-04-08 by terrypacker