MangoAutomation/ma-dashboards #195: Script output areas too small

Puckfist, 2019-04-01:
The script output areas are both too small and mislabeled as being "Error console" for email handlers and set point scripts. It is script output and can be useful for script writing to invoke print functions on the context objects provided (which they support and provided a print function for)

jazdw, 2019-04-02:
Looks like @Craig-IAS changed the translation here a727f60737ee39e78baf2f21054fd3d33bd2cdce

@guetteluis can you fix up the other things as this is a component you created. Please make the component resizeable via a resize handle in the bottom right/bottom edge. See ma-ace-editor (although that one seems to be behaving quite strangely).

luisguette, 2019-04-03:
fixed on https://github.com/infiniteautomation/ma-dashboards/commit/1ec9a6d8294d8012ee6f4688fdc32e2873207c9b
[image]

jazdw, 2019-04-04:
@luisguette I just took more of a look at this component
- Why is there text that says "Script output" that disappears on validation? This should be removed IMO
- There needs to be a margin above the log level drop down
- There is wasted space between the script output and the validate button, remove it
- I saw a couple of md-colors attributes like this `md-colors="{background: 'grey-900'}"`, we should not use a specific color palette like 'grey', instead use 'primary', 'accent', 'warn' or 'background' so the user has control over them. As a side note, you can use a [one time binding](https://docs.angularjs.org/guide/expression#one-time-binding) for these static color expressions e.g. `md-colors="::{color: 'warn'}"`

luisguette, 2019-04-04:
great, I will work on that

luisguette, 2019-04-04:
fixed
[image]

jazdw, 2019-04-04:
Thanks, but the initial issue was that the output was too small. We seem to have regressed?

luisguette, 2019-04-05:
[image]

It has a max-height of 300px, but I can make it bigger

jazdw, 2019-04-05:
@luisguette OK, got it, it expands with the text content up to `max-height`. Can you please change `overflow-y: scroll` on the div to `overflow-y: auto`, the screenshots above do not represent the current code.

I think it would also put another button underneath "Validate" that closes the script output.

luisguette, 2019-04-05:
ok, I will do that

guetteluis, 2019-04-05:
done in https://github.com/infiniteautomation/ma-dashboards/commit/f014b2beae1d67cc07d859e4c87c4e4e37a7826f

jazdw, 2019-04-08:
@guetteluis Please open a PR for your branch.

As I said above, please put the close button below "Validate" (make it raised and just say "Close output").

Other things that need to be fixed -
- Missing translation as shown in picture
- When closing the output the line stays highlighted and is still marked as having an error

[image]

jazdw, 2019-04-09:
Thanks @guetteluis

closed 2019-04-09 by jazdw