MangoAutomation/ma-dashboards #197: Edit multiple data points, chart renderer tab has issue

Puckfist, 2019-04-03:
Typically when editing multiple points <<multiple values (n)>> seems displayed in fields where the points don't already have the same setting. With the chart renderer tab, however, I am seeing the selector for chart renderer type disabled with that value when they do not have the same chart renderer, but blank and disabled when they do:

[image]

jazdw, 2019-04-03:
I cant reproduce this. Judging by your picture you are trying to edit two numeric data points which both currently have a statistics chart renderer?

Craig-IAS, 2019-04-04:
If I select a Binary point and an alphanumeric point, click edit and try to change their device name.  I cannot save. The chart renderer type is disabled and empty so the form validation does not allow you to save.
[image]

jazdw, 2019-04-04:
Thanks @Craig-IAS that helped identify it.

commit 1a72ab89 (2019-04-04): Fix chart renderer options with multiple data types

Closes #197
closed 2019-04-04 by jazdw