MangoAutomation/ma-dashboards #199: Edit multiple data points of different data type

Craig-IAS, 2019-04-04:
# 1 If I select a binary point and then select a numeric point, click edit and then save without changing any properties both points have an error.

# 2 If I select a numeric point and then a binary point and do the same only one has an error. The first selected point does not error

after comparing & inspecting the API requests with chrome I noticed the following. 
#1 API request:
The binary point had the discrepancies in the loggingProperties and chartRenderer:
```javascript
 "loggingProperties": {
                "loggingType": "ON_CHANGE",
                "tolerance": 0,
                "discardExtremeValues": false,
                "overrideIntervalLoggingSamples": false,
                "cacheSize": 1,
                "intervalLoggingPeriod": {}
            },
```

```json
"chartRenderer": {
                "limit": 10,
                "type": "chartRendererTable",
                "timePeriod": {}
            },
```
I dont believe the timePeriod should be in the chartRenderer object(not sure if the API will error with this)
 I dont believe the intervalLoggingPeriod should be in the loggingProperties object(not sure if the API will error with this).

The numeric point had discrepancies in the loggingProperties and chartRenderer and textRenderer:
```json
"loggingProperties": {
                "loggingType": "INTERVAL",
                "tolerance": 0,
                "discardExtremeValues": false,
                "overrideIntervalLoggingSamples": false,
                "cacheSize": 1,
                "intervalLoggingPeriod": {}
            },
            "textRenderer": {
                "type": "textRendererAnalog"
            },
            "chartRenderer": {
                "type": "chartRendererImage",
                "timePeriod": {}
            }
```
No data in intervalLoggingPeriod, missing data in the textRenderer and timePeriod. 

#2 API request:

Numeric Point did not error and loggingProperties,textRenderer and chartRenderer were complete.
```json
loggingProperties": {
                "loggingType": "INTERVAL",
                "intervalLoggingType": "AVERAGE",
                "intervalLoggingPeriod": {
                    "periods": 1,
                    "type": "MINUTES"
                },
                "tolerance": 0,
                "discardExtremeValues": false,
                "overrideIntervalLoggingSamples": false,
                "cacheSize": 1
            },
            "textRenderer": {
                "useUnitAsSuffix": false,
                "format": "0.00",
                "suffix": "",
                "type": "textRendererAnalog"
            },
            "chartRenderer": {
                "timePeriod": {
                    "periods": 1,
                    "type": "DAYS"
                },
                "type": "chartRendererImage"
            }
```

Binary point did error 
```json
"loggingProperties": {
                "loggingType": "ON_CHANGE",
                "intervalLoggingPeriod": {},
                "tolerance": 0,
                "discardExtremeValues": false,
                "overrideIntervalLoggingSamples": false,
                "cacheSize": 1
            },
            "textRenderer": {
                "type": "textRendererBinary"
            },
            "chartRenderer": {
                "timePeriod": {},
                "type": "chartRendererTable"
            }
```

Hope my debugging narrows done the search

jazdw, 2019-04-04:
OK, think I have this one sorted, this is the commit 56b0659e21364b9deecb9cc70d504586555a2ced
Can you please test it out.

It was due to the way we combine multiple points into a single object then split them back into individual ones when saving. Basically if one point has a property like `"timePeriod": {"periods": 1,"type": "DAYS"}` and the other doesn't have the "timePeriod" property when we were converting back to an array of points the point ended up with a property like `"timePeriod": {}` which failed to validate on the server.

Craig-IAS, 2019-04-07:
I tested all the use cases I could come up with and have not got an error again. I believe this can be closed.

closed 2019-04-07 by Craig-IAS