MangoAutomation/ma-dashboards #201: Adding BACnet points via the scan tool doesn't allow have selection for name field

Joelhaggar, 2019-04-05:
On the legacy UI for BACnet you can select to use the Object Name or Description as the Mango Data Point name.  This is not available in the new UI scan tool and looks like it is defaulting to use the Description.

In this example, the device has a Name and Description but most of the points get populated with "Default Bacnet Description" as their name
[image]

This device populates the Mango names correctly:
[image]

jazdw, 2019-04-05:
Yeah I was considering just making it add both the name and description as tags. It currently uses the description if it is not empty and falls back to the name.

"Default Bacnet Description" must be coming from either the BACnet4J library or the device itself. Its not something in the UI.

terrypacker, 2019-04-06:
"Default Bacnet Description" would be coming from the device you are exploring that is simply passed through from the message received in BACnet4J --> REST API --> UI

jazdw, 2019-04-10:
Closed via infiniteautomation/ma-modules-private@c96d468a7d79fb50baf22d0365bea47e0526b7c5

commit 131b355c (2019-04-09): Add help page re. point templates

Closes #201
closed 2019-04-10 by jazdw