MangoAutomation/ma-dashboards #203: Bulk data point editor: display message if point has a Template

terrypacker, 2019-04-09:
We need to display a message when editing a point with a template applied.  Ideally we will also disable/hide the inputs that are affected by the template being applied upon save.

jazdw, 2019-04-09:
@terrypacker I thought the consensus was to remove the template upon save?

terrypacker, 2019-04-09:
@jazdw you are correct but I think we also decided to show a message if you start editing a point or points that have templates applied so that it is obvious that they will be removed.

jazdw, 2019-04-09:
[image]

Joelhaggar, 2019-04-09:
I wonder if we should update this page and give a link to is so they can learn about what a Data Point template is https://help.infiniteautomation.com/using-data-point-templates or maybe there is a help page in Mango we can use for this.  I think many users might not know what this is as using a template was the default when saving a data point in the past.

terrypacker, 2019-04-09:
There is a 20 line help file at WEB-INF/dox/dataPointTemplate.htm in the core.

jazdw, 2019-04-09:
@Joelhaggar @terrypacker Please check help content and edit it if necessary https://github.com/infiniteautomation/ma-dashboards/blob/main/UI/web-src/ui/views/help/dataPointTemplate.html

closed 2019-04-09 by jazdw

terrypacker, 2019-04-10:
@jazdw we have both reviewed it and it is fine.  Thanks.

commit ecb306b0 (2019-04-09): Remove templateXid when saving points

Closes #203
commit 45365fc8 (2019-04-09): Add help page re. point templates

Closes #203