MangoAutomation/ma-dashboards #206: Can't edit an "Internal" DS as the "Create points pattern" is empty and is marked as required.

jazdw, 2019-04-09:
Can't edit an "Internal" DS as the "Create points pattern" is empty and is marked as required.

[image]

Craig-IAS, 2019-04-09:
This seems to only happen on the default internal data source. If you create a new internal data source it does not happen.
Also if you look at the API response when opening the data source page.  The default internal data source has :   "createPointsPattern" : null,

jazdw, 2019-04-09:
Well shit. @terrypacker can you look at this.

terrypacker, 2019-04-09:
@jazdw and @Craig-IAS This is a UI bug.  It is totally valid for this data source to have a null create points pattern.  This should correspond to 'NONE' in the drop down.  My guess is that when loading the data source you are not setting the drop down value for the create points pattern.

You can reproduce this by creating a new data source and setting the create points pattern to NONE then reload it and try saving it.  Same problem.

Craig-IAS, 2019-04-10:
My bad for leading us astray.  Fixed here: 
https://github.com/infiniteautomation/ma-modules-public/commit/3151f2f0c02c900110cd52afa8975180a24945eb

closed 2019-04-10 by Craig-IAS