MangoAutomation/ma-dashboards #207: Meta point script does not validate if using the point's own variable

jazdw, 2019-04-10:
Meta point script does not validate if using the point's own variable. See screenshot. I assume this is because its not added to the array of context points when sending it to the validation endpoint. Please test with a new data point and when updating an existing data point.

[image]

guetteluis, 2019-04-10:
When the data point is new, I disabled the Validate button and added a warn message at the top of the script editor. What do you think @jazdw ?
[image]

jazdw, 2019-04-11:
@guetteluis Please reference the commit so I can see what you changed.

guetteluis, 2019-04-11:
I added a `disable-validation` property for scripting editor component in https://github.com/infiniteautomation/ma-dashboards/commit/6e40afc4809a510d12800d41e1b7e787106578d4 and added the `saveFirst` message in https://github.com/infiniteautomation/ma-modules-private/commit/7e68989e20c33a9b13d5459f11eedc773e968e67

jazdw, 2019-04-11:
OK looks good please create PRs.

terrypacker, 2019-04-11:
@guetteluis I am currently editing a previously saved point and getting this error:

```
'ReferenceError: "my" is not defined' on line 1
ReferenceError: "my" is not defined
```

Please ensure that the editing point is added to the array of context points.  I have not tested this on your branch but neither of the commits referenced above will fix what I am seeing.

closed 2019-05-07 by terrypacker