MangoAutomation/ma-dashboards #214: Excel Report Template - No error displayed when template file DNE

terrypacker, 2019-04-11:
I was trying to edit an Excel Report Template that I imported but the excel template file was not brought across.  The page does nothing but the log output is:

```
DEBUG 2019-04-11T10:13:36,457 (com.infiniteautomation.mango.rest.v2.exception.RestExceptionHandler.handleExceptionInternal:184) - Rest API Exception 
com.infiniteautomation.mango.rest.v2.exception.GenericRestException: Report template file not found
```

jazdw, 2019-04-11:
@terrypacker please check the REST response and let me know what the REST response contains.

terrypacker, 2019-04-12:
[image]

The JSON payload is:
```
{
  "cause" : null,
  "mangoStatusCode" : -1,
  "mangoStatusName" : null,
  "localizedMessage" : "Report template file not found"
}
```

terrypacker, 2019-04-12:
I could make the response a 404 but that is a little ambiguous if you don't look at the message as the actual template VO may not exist.

terrypacker, 2019-12-12:
@guetteluis when you start working on Excel Reports I'd like to fix this one up too.

guetteluis, 2020-01-22:
Hey @terrypacker can you explain me more about how do you get this error? Thanks

terrypacker, 2020-01-30:
@guetteluis you should create an Excel Report Template then go to the file store where the template is and delete the file.  Then try to open the template and edit it.

closed 2020-02-18 by terrypacker