MangoAutomation/ma-dashboards #218: Data Source toggle button changes not reflected in edit view

terrypacker, 2019-04-12:
I know @guetteluis mentioned this in a PR but I'm adding as an issue so its not forgotten.  When editing a data source if you toggle the source via the toggle in the list the change is not reflected in the edit view.  This means when you save it the state of the data source may change to what is reflected in the list.

I know this is sort of like the websocket issues you closed but in this case I think that input should at least change with the state of the input on the same page.

jazdw, 2019-04-13:
Copying my response from infiniteautomation/ma-modules-public#40
> When you start editing a DS (or any other item in the new UI) it creates a copy. You then edit this snapshot in time of the DS, Enabling or disabling a DS via the list on the left takes place immediately. In short what you are seeing is the intended behavior.

This has been bought up many times, by you, Luis, Phillip and Joel. It seems like it is still confusing people despite the fact that I put an extra comment next to the enable button. I still believe we need to implement versioning/time-stamping of VO columns in the DB to address this issue.

terrypacker, 2019-04-15:
It feels non intuitive the way it is now.  I don't like the fact that the 2 toggle buttons can get out of sync.  Is there really a need for there to be an input for Enabled in the edit view since that can be done in the select list for each data source?

jazdw, 2019-04-15:
> Is there really a need for there to be an input for Enabled in the edit view since that can be done in the select list for each data source?

Yes, the data source editor is a component that is can be used outside of the data source edit page. As "enabled" is a property of the data source it makes sense for there to be an input to change it.

guetteluis, 2019-04-15:
@terrypacker @jazdw  I was thinking, that the toggle in the DS list can be hidden when the DS is selected

jazdw, 2019-04-30:
I've added a possible work around above, along the lines of what @guetteluis suggested but I'm disabling the switch. Can you guys please test it and see what you think. Perhaps @Joelhaggar too.

Also @terrypacker another reason we need a switch inside the editor component is for the small view (try making your browser window small).

I guess we could provide an attribute to hide the enabled switch? That way anyone who uses the component can still have the enabled switch and we can show it for small windows.

We would then copy the current enabled status to the DS before saving it. This could be finicky though, imagine if you lost connection for a few seconds and missed the WS update saying the DS is now disabled. You could then end up inadvertently changing the DS back to enabled.

Ultimately this holds true:
> I still believe we need to implement a version/time-stamp column in VO tables in the DB to address this issue.

terrypacker, 2019-05-09:
I like it.  Agreed about database versioning, but for now this is a good solution.  I'm happy to close this issue at anytime now.

jazdw, 2019-06-28:
Closed via 48f345af121b0004ceaec035f02de7e6ace19b4c and 1325fb9f69820b126ab05504413c71292fb70e6a

The switch is now hidden completely when editing the DS.

closed 2019-06-28 by jazdw