MangoAutomation/ma-dashboards #224: BACnet Local Device - Delete button inactive

terrypacker, 2019-04-17:
When I go to delete a previously created BACnet local device the 'Delete' button is greyed out.  I must edit an input in order for that button to become active.  

The button should be active when I select an already created local device in case I want to delete it.

guetteluis, 2019-04-17:
hey @terrypacker I couldn't reproduce this error, can you be more specific? thanks

terrypacker, 2019-04-17:
@guetteluis I figured it out.  We may need to think about this, the `Device name` input is not required by the backend validation so if you save a local device without a device name you can get into this situation where the list is all messed up.

I found this by using some node tests that create local devices without a device name but you can still save it in the UI without a name.

terrypacker, 2019-04-17:
There is no reason in the backend to have a device name set, its only for display.  If we change that we will have to think about the ramifications of JSON imports from legacy systems.  If it is easy in the UI I would suggest that we just allow that field to be empty.  But it seems like there is something in the list logic that needs that set...

guetteluis, 2019-04-17:
oh ok, let me check

closed 2019-05-09 by terrypacker