MangoAutomation/ma-dashboards #225: BACnet Local Device - Delete last local device

terrypacker, 2019-04-17:
There seems to be some bugs around the select list for the devices and the editing UI.  I just created 3 local devices and then tried to delete all of my local devices one by one.

When deleting the last item in the list I got this in ma.log:

```
WARN  2019-04-17T06:20:34,980 (com.infiniteautomation.mango.rest.v2.ServerRestV2Controller.postClientError:318) - Client error
[user=admin, cause=Possibly unhandled rejection: {}, location=http://mango.terrypacker.com:8081/ui/administration/bacnet-local-devices/e9af781d-f76a-435f-91f8-0175c62f69c1, userAgent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_14_4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/73.0.3683.103 Safari/537.36, language=en-US, date=2019-04-17T06:20:34.972-10:00, timezone=Pacific/Honolulu]
TypeError: Cannot read property '0' of undefined

guetteluis, 2019-04-18:
@terrypacker I found the error and fixed it in: https://github.com/infiniteautomation/ma-modules-private/commit/7c04f9c9172f914be6714b5ff931834839fb68b3

But this leaves me a doubt. Is it correct that it is `id` and not `xid`? probably we talked about this before, but I don't remeber.

terrypacker, 2019-04-18:
@guetteluis no, you are correct.  Local devices are the only non-standard exportable object that doesn't have an XID.  It only has an ID.

closed 2019-05-09 by terrypacker