MangoAutomation/ma-dashboards #234: Display millisecond event durations accurately

terrypacker, 2019-04-18:
I am testing a bug in the event detectors and know that the time between my events being raised and RTN is around 2ms.  The UI is misleading in telling me that it is a few seconds.

Ideally if the duration is < 1s you could display the number of milliseconds.

[image]

jazdw, 2019-04-19:
Currently we use this - https://momentjs.com/docs/#/durations/humanize/
If you want me to change this, you need to give me the formats you desire for all the different duration lengths.

jazdw, 2019-04-30:
@terrypacker any input on the above?

terrypacker, 2019-04-30:
@jazdw sorry I looked into it but didn't see any options for humanize to go down to the millisecond level accuracy?

jazdw, 2019-04-30:
@terrypacker That is what we currently use, I'm saying if we aren't going to use that method we need to specify what formats we are going to use and for what durations. i.e.
[0, 1) second => {0} ms
[1-60) seconds => {0} s
[60-3600) seconds => {0} m {1} s
etc.

terrypacker, 2019-04-30:
Gotcha.

The general idea was that when tracking down why an event is being raised it can be nice to know some exact info on how long it was active.  But generally we won't care too much on the accuracy.

Maybe we want to make that clickable and on-click it shows the duration accurate to the ms?

```
[0, 1) second => {0} ms
[1-60) seconds => {0} s {1} ms
[60-3600) seconds => {0} m {1} s {2} ms
[1hr-23h59m59s) => {0} h {1} m {2} s {3} ms
[1 day-forever] => {0} days {1} h {2} m {3} s {4} ms
```

jazdw, 2019-05-02:
Don't really want to make it clickable, and it has to be succinct. The number of milliseconds is pretty superfluous once you get past a few seconds don't you think? How about for now I just make it display the "x ms" for anything under 1 second and use humanize for everything else?

terrypacker, 2019-05-02:
Sure that's fine. If anyone else ever brings it up we can revisit it then.

jazdw, 2019-05-02:
Made it so that
Less than 1 second displays x milliseconds
Less than 5 seconds displays x.x seconds (1 decimal place)
Less than 60 seconds displays x seconds (0 decimal place)
Everything else uses humanize()

commit 391ea4f1 (2019-05-02): Display small durations more accurately

Closes #234
closed 2019-05-02 by jazdw