MangoAutomation/ma-dashboards #237: New tags don't show up in bulk editor

terrypacker, 2019-04-26:
I was testing a brand new system with 0 tags in it.  I added a tag but it didn't show up in the 'Select tags' drop down until I refreshed the page.

Steps:
 1. I went to the bulk editor, selected Devices
2.  Select 1 point for editing
3.  Add tag to that point and save
4.  Returned to bulk editor and no new tag in drop down

This seems to be the same whether there are pre-existing tags in the system or not.

jazdw, 2019-04-29:
Screenshot please. I don't know which "Select tags" drop down you are referring to. To confirm, you are talking about 3.6.0 yes?

terrypacker, 2019-04-29:
Yes Mango 3.6.0 latest build.

[image]

terrypacker, 2019-04-29:
The test tag drop down in the upper right of that screenshot.  It doesn't obtain the new tags until the page is refreshed.

commit d514992a (2019-04-30): Refresh available tags when saving data points

Closes #237
closed 2019-04-30 by jazdw