MangoAutomation/ma-dashboards #238: Process event handlers "changeing to type email"

Craig-IAS, 2019-04-29:
**Describe the bug**
On the event handler page.  If you create or work on an event handler that is not an email, save it, then click on "Create new event handler".  Then click on an existing event handler the existing event handler is of type email and all the tabs are for an email EH.

**To Reproduce**
Steps to reproduce the behaviour:

1.  Go to event handler page and create a process event handler and save it. 
2. Click on the create a new event handler button.
3. Do not do anything on the new EH.
4. Click on the EH that was made in step one.
5. It will be of type Email and all associated tabs will be for an email EH.

**Expected behaviour**
I would expect that when clicking on an existing EH that is not an email EH to be able to edit it. 
When refreshing the page the issue no longer exists. 

**Screenshots**
Before:
[image]
After:
[image]

**Mango Version (please complete the following information):**
 - Core Version: 3.6 nightly build 29th April

**Browser (please complete the following information):**
 - OS: linux mint
 - Name chrome
 - Version 22

**Additional context**
When I refresh the page and all the EH types that I have created are correct.  The only way I can get the bug to come back is to save a process EH **while in the process tab, and click on the create event handler button while still in the process tab** 

**ma.log Snippet**
- Check your ma.log file (located in ${MA_HOME}/logs) for any errors or warnings
- Do NOT post the entire file, just the relevant lines

**Browser console errors**

none

Craig-IAS, 2019-04-29:
This is not limited to process event handler

Craig-IAS, 2019-04-29:
Starting to find a lot of different ways to reproduce this.

Craig-IAS, 2019-04-30:
if you save an event handler,  then try to reselect it in the list.  It is an email event handler

jazdw, 2019-05-14:
@terrypacker I think this is a WebSocket issue, there is no `handlerType` property.

[image]

terrypacker, 2019-05-14:
@jazdw I think we will need to both look at this.  The handlerType is being written out when we make a REST GET request but not when we are sending the websocket message.  I have verified that both are using the correct ObjectMapper but it seems that in the websocket since we are creating a writer and executing `writer.writeValueAsString(userMessage)` [here](https://github.com/infiniteautomation/ma-modules-public/blob/d6143ae7add194fda3a6ecbc9fa349ab3e148e08/Mango%20API/src/com/infiniteautomation/mango/rest/v2/websocket/DaoNotificationWebSocketHandler.java#L76) that call is not aware of the JsonTypeInfo for the event handler subtypes.

I can fix this by explicitly returning the handlerType in a getHandlerType() method.  That is how the data sources/publishers etc work.  However I think it is worth investigating the reason this is happening before choosing the easy fix.

terrypacker, 2019-05-14:
A little more info.  The handlerType is known during serialization because we register the subtypes via the RestModelJacksonMapping class which is used both for the RestModelMapper and Jackson serialization.

terrypacker, 2019-05-14:
For now I just added the method so the UI will work.  Will have to look into it more at some point in the future.

jazdw, 2019-05-15:
Just had a dig into it and it seems that it is because Jackson doesn't know the type of the object it is serializing, we just pass an Object to `writeValueAsString()`. If you pass an `AbstractEventHandlerModel` to `writeValueAsString()` it works fine, I think something like a `List<AbstractEventHandlerModel>` would also work. So yeah I think the fix you made is fine. Thanks.

closed 2019-05-15 by jazdw