MangoAutomation/ma-dashboards #241: BACnet Error Not Displayed when adding point from Device Browser

terrypacker, 2019-05-02:
I just found that we previously did not support the Multistate Value BACnet Object type.  When I tried adding that from a discovered device the API response with a 400 Bad Request but nothing is displayed in the UI.

REST Response from GET `http://mango.terrypacker.com:8081/rest/v2/bacnet/object-properties/MULTI_STATE_VALUE`

```
{
  "cause" : null,
  "mangoStatusCode" : 4005,
  "mangoStatusName" : "BAD_REQUEST",
  "localizedMessage" : "Bad Request"
}
```

Backend stack trace:
```
DEBUG 2019-05-01T14:22:18,260 (com.infiniteautomation.mango.rest.v2.exception.RestExceptionHandler.handleExceptionInternal:184) - Rest API Exception 
com.infiniteautomation.mango.rest.v2.exception.BadRequestException: Bad Request
	at com.infiniteautomation.mango.rest.v2.BACnetRestController.getPossibleProperties(BACnetRestController.java:923) ~[?:?]
```

I'm fixing this to return a message about that data type not being supported but it will still throw a BadRequest exception (unless you prefer another type).

To reproduce you can setup a Mango BACnet publisher to publisher a Multistate point as a Multistate value type and then explore it.  Or easier yet, just modify the REST controller to always throw a BadRequest exception.

terrypacker, 2019-05-02:
Ok, I dug into this a little more and I see what is happening.  The error message is not being displayed but we actually do support Multistate Value object types.

So... when the `BACnetPointLocatorVO.OBJECT_TYPE_CODES` was created the formatting was done differently for some of the codes.  For example the BACnet4J ObjectType name for multi state input is `multi-state-input` but the Point Locator export code is `MULTISTATE_VALUE`.  You must be doing some massaging in the UI to format `multi-state-value` into `MULTI_STATE_VALUE`.  

I think I have a much cleaner approach in the REST controller which is to have you send in the "lowercase with dashes" object type and have me find out if that fits any supported types we have.

I see that there is also a bug in the UI where you would fail to get the object type from the objectTypes array in the bacnet.js file for this (and several other) object type.

terrypacker, 2019-05-02:
I've modified the endpoint to accept both the lowercase with - format AND Mango Export Codes.  It now will return more information in an Object rather than just list of properties that slots in nicely with the UI code:

Here is the response:
```
{
  "objectType" : {
    "javaType" : "com.serotonin.bacnet4j.type.enumerated.ObjectType",
    "id" : 19,
    "value" : "multi-state-value"
  },
  "typeName" : "MULTISTATE_VALUE",
  "translation" : "Multi-state value",
  "properties" : [ {
    "propertyName" : "acked-transitions",
    "propertyId" : 0,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "event-time-stamps",
    "propertyId" : 130,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "alarm-values",
    "propertyId" : 7,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "notify-type",
    "propertyId" : 72,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "number-of-states",
    "propertyId" : 74,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "object-identifier",
    "propertyId" : 75,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "object-name",
    "propertyId" : 77,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "object-type",
    "propertyId" : 79,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "out-of-service",
    "propertyId" : 81,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "notification-class",
    "propertyId" : 17,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "present-value",
    "propertyId" : 85,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "priority-array",
    "propertyId" : 87,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "description",
    "propertyId" : 28,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "event-message-texts",
    "propertyId" : 351,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "event-message-texts-config",
    "propertyId" : 352,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "event-detection-enable",
    "propertyId" : 353,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "event-algorithm-inhibit",
    "propertyId" : 354,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "event-enable",
    "propertyId" : 35,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "event-algorithm-inhibit-ref",
    "propertyId" : 355,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "event-state",
    "propertyId" : 36,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "time-delay-normal",
    "propertyId" : 356,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "reliability-evaluation-inhibit",
    "propertyId" : 357,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "profile-location",
    "propertyId" : 485,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "tags",
    "propertyId" : 486,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "reliability",
    "propertyId" : 103,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "fault-values",
    "propertyId" : 39,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "relinquish-default",
    "propertyId" : 104,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "profile-name",
    "propertyId" : 168,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "state-text",
    "propertyId" : 110,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "command-time-array",
    "propertyId" : 430,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "status-flags",
    "propertyId" : 111,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "current-command-priority",
    "propertyId" : 431,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "last-command-time",
    "propertyId" : 432,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "time-delay",
    "propertyId" : 113,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  }, {
    "propertyName" : "value-source",
    "propertyId" : 433,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "value-source-array",
    "propertyId" : 434,
    "supportedDataTypes" : [ "ALPHANUMERIC" ]
  }, {
    "propertyName" : "property-list",
    "propertyId" : 371,
    "supportedDataTypes" : [ "ALPHANUMERIC", "BINARY", "MULTISTATE", "NUMERIC" ]
  } ]
}
```

jazdw, 2019-05-02:
Display error message - infiniteautomation/ma-modules-private@3ceb06edae99b88f68cbb901c868c66e004e6af3

jazdw, 2019-05-02:
@terrypacker infiniteautomation/ma-modules-private@edf0beb25761632d9e564f804a3428b9373b8d6f
This should fix up the browser point import. Re. translations, not sure where you wanted to use this as the REST end point that we were looking at does not allow getting a list of object types.

closed 2019-05-02 by jazdw