MangoAutomation/ma-dashboards #243: Closing Data Point Purge modal causes REST exception

terrypacker, 2019-05-04:
When testing purging point values I found that an Exception is being thrown by the REST api when the UI tries to cancel an already completed purge.  

To reproduce:

1. Edit a data source
2. Select a point (check its box)
3. Click the purge trashcan
4. Check purge all data for selected point
5.  Click PURGE POINT VALUES
6.  Click the x to close it after the purge finishes

See the DEBUG message in the backend as a PUT to v2/point-values/purge/{id} is made to cancel the resource but its already completed.

[image]

terrypacker, 2020-04-03:
@pierpuccini see if this is still happening.

pierpuccini, 2020-04-06:
No errors occuring in debug console, and mentioned PUT request is now a POST.

closed 2020-04-06 by pierpuccini