MangoAutomation/ma-dashboards #246: Progress Bar Updates - Not fluid (perhaps blocking code?)

terrypacker, 2019-05-06:
I am testing the Meta Data Source history generation and found that the websocket messages are being sent to the UI regularly to update the status of the generation.  However the progress bar is only updated periodically and unrelated to the websocket message being received.  

The UI code uses the TemporaryRestResource class to receive the updates and display them.  

Something interesting to note is that when debugging this in Chrome Dev Tools I see the progress bar get updated after each successful (periodic) call to get the current user:

`GET /rest/v1/users/current`

It seems like the page is waiting for that call to complete before it updates the progress bar with the information received in the web sockets.

jazdw, 2019-05-14:
It sounds like the websocket messages are not being processed inside the AngularJS digest cycle. We probably are missing a $scope.$apply(). Can you post a screen shot and/or a link to the page where this code is.

jazdw, 2019-05-14:
OK found the code, these are the pertinent files.
https://github.com/infiniteautomation/ma-modules-private/blob/main/Meta%20Data%20Source/web-src/components/metaDataSourceEditor/metaDataSourceEditor.js
https://github.com/infiniteautomation/ma-modules-private/blob/main/Meta%20Data%20Source/web-src/services/metaDataSource.js

@guetteluis You extended TemporaryRestResource then didn't use any of its features for temporary resources... Please look at how this class is used elsewhere (e.g. purging point values). You should
1. Create an instance of the TemporaryRestResource e.g. `const task = new maMetaHistoryGeneration({options...})`
2. Start the task / temporary resource e.g. `const promise = task.start($scope)`
3. Use the promise progress callback e.g. `promise .then(done => {...}, error => {...}, progress => {...})`

guetteluis, 2019-05-15:
> OK found the code, these are the pertinent files. [...]

@jazdw I will use the `start($scope)` method. But, as @terrypacker and I saw, if a user enter to the view, and a generation is already running, It should show the progress of the current generation through the websocket connection, and you will see that the progress bar updates are not fluid.

jazdw, 2019-05-15:
> if a user enter to the view, and a generation is already running, It should show the progress of the current generation through the websocket connection

OK, yes so we need to account for this. So the solution will be slightly different. We may need to add a method to the maTemporaryRestResource for this, I will take a look.

> you will see that the progress bar updates are not fluid.

As I said in my previous comment this is due to you using WebSocket messages outside of the digest cycle.

jazdw, 2019-05-15:
@guetteluis I'm going to modify the REST controller a bit so just leave this with me fro the moment.

guetteluis, 2019-05-15:
ok

jazdw, 2019-05-16:
OK so in addition to the commit referenced above these commits fix this issue. However I am not happy with the way it currently displays the results and error messages so I will continue to work on this.

infiniteautomation/ma-modules-private@bc6aae96f3e091fa4d256cb29da5a9680ebe2324
infiniteautomation/ma-modules-private@3a99987ddff477ea912a1fac3caa56d4bb150cf4
infiniteautomation/ma-modules-private@f8ea8ac6c39af6a270af13f260a8353758ed210b

jazdw, 2019-05-16:
@guetteluis Please review the above commits so you can see how much simpler it is now.

closed 2019-05-16 by jazdw