MangoAutomation/ma-dashboards #251: Translations are not showing up sometimes when use maTr filter

guetteluis, 2019-05-08:
When `label="{{'translation.key' | maTr}}"` is used, sometimes it show the translations, and sometimes it doesn't

you can check:

- Set point event handler component
- Script context component

jazdw, 2019-05-14:
@guetteluis you should generally avoid using the `maTr` filter wherever possible. This is because AngularJS filters cannot be asynchronous so I cannot load the translations from the REST API before translating the key. The filter only works for namespaces that are already loaded. So the options are -

1. Dont use `maTr` filter, use the `ma-tr` directive instead for example or use the `maTranslate` service. If you give me an example of what exactly you are using `maTr` for I can provide an alternative
2. Use a translation namespace that is already loaded e.g. `ui`
3. Pre-load the namespace using a menu item / state's resolve option

guetteluis, 2019-05-14:
@jazdw oh, I see. So, in this case:

```
<div flex="100">
    <ma-filtering-point-list auto-init="false" label="{{'dsEdit.script.scriptContext' | maTr}}" name="contextPoint" ng-change="$ctrl.selectContextPoint()" ng-model="$ctrl.contextPoint"></ma-filtering-point-list>
</div>
```

I think that the best option would be to use some translation key like `ui.script.context`, right?

guetteluis, 2019-05-15:
I've updated the script context translation here: https://github.com/infiniteautomation/ma-dashboards/pull/258/commits/58f875e3cfbfe52ab01ea2c8f9939cac7052d763

closed 2019-06-26 by jazdw