MangoAutomation/ma-dashboards #261: Virtual Serial Port - add inputs for ip white list

terrypacker, 2019-06-19:
The backend supports adding a list of ip addresses that can connect to the open port of a Virtual Serial Port (Server Socket) but there are no inputs in the UI for this.

Add the inputs and help to the system settings section.

The the model/backend has all the validation and getter/setter implementation.

```
{ ...
    ipWhiteList: [String, String, ... ],
```

guetteluis, 2019-08-06:
@terrypacker I have added the IP white list input, but I don't see the parameter in the model from the backend. I have updated `mangoAPI` and the `core`.

terrypacker, 2019-08-19:
They definitely are there.  This is only available on the Serial Server Socket Bridge:

[image]

jazdw, 2019-09-05:
@guetteluis Any updates on this?

luisguette, 2019-09-05:
@jazdw I haven't had time, but I can work on these issues tomorrow.

terrypacker, 2020-02-25:
@pierpuccini can you test and confirm this is working.  Basically add a few whitelisted IPs and then confirm they are saved and reloaded in the UI

pierpuccini, 2020-02-29:
Is this how the test should appear? If it is, I added my own IP, saved, refreshed and it appeared.

[image]

closed 2020-03-03 by terrypacker