MangoAutomation/ma-dashboards #265: Binary Starts and Runtimes on Chart

terrypacker, 2019-06-20:
This is where we can chart a binary point as run time, starts, etc. 

Ok, yep thanks.  So if you use the Rollup Type of ALL for a binary point you can get the startsAndRuntimes list:
```
    {
        "timestamp": "2018-01-01T00:00:00.000+00:00",
        "start": null,
        "first": {
            "timestamp": "2018-08-21T21:20:17.847+00:00",
            "value": false
        },
        "last": {
            "timestamp": "2018-08-28T00:34:57.092+00:00",
            "value": true
        },
        "count": 58556,
        "startsAndRuntimes": [
            {
                "value": false,
                "starts": 29278,
                "runtime": 312608873,
                "proportion": 0.027387279382251817
            },
            {
                "value": true,
                "starts": 29278,
                "runtime": 11101773280,
                "proportion": 0.9726127206177482
            }
        ]
    }
```
An example URL would be: http://{{hostname}}:{{port}}/rest/v2/point-values/time-period/binary/ALL?from=2000-01-01T00:00:00.000Z&timePeriodType=YEARS&timePeriods=1&dateTimeFormat=yyyy-MM-dd%27T%27HH:mm:ss.SSSxxx&limit=100

jazdw, 2019-06-26:
@terrypacker what is this about? I have no idea.

terrypacker, 2019-06-27:
There has been a longstanding request by @Joelhaggar to plot this information, I just copied it into git.  I'm not exactly sure how it would work, I guess as a bar chart or something.  Maybe he can elaborate?