MangoAutomation/ma-dashboards #275: Add system setting to control which tags are displayed for data point label

jazdw, 2019-07-02:


jazdw, 2019-07-02:
I think probably all tags should be displayed by default, the systems setting would probably just control which tags to exclude. Add a formatLabel() function to the DataPointVO. I use a similar method in the front end which displays the label as something like `Dashboard Demo — Multistate [site=My site]`. We should update all the Java code in the backend which prints a point label to use our new method e.g. reports (I think we did a bit of work on this already).

terrypacker, 2020-02-20:
I have added a system setting `tags.dataPoint.display` that holds a comma separated list of tag keys.  Currently there is limited validation to ensure they are not name or device name.  Also do we want to make this respect the order of the tags in this setting?

The data point model now has this in it based on the tags in the system setting

` "formatLabel" : "Voltage - Meter 1 [phase=Phase C, site=Longmont]",`

terrypacker, 2020-02-20:
@jazdw this is ready for you to use

jazdw, 2020-02-24:
> Also do we want to make this respect the order of the tags in this setting?

Absolutely, fixed here infiniteautomation/ma-core-public@cb7f06f5cf5bb3cb088b7496a1bbe4172c71aadf

The DataPointModel property is now extendedName
infiniteautomation/ma-modules-public@703624fa51342f5dd6f810a90f7cb8a94269b7ce

System setting UI added here 74dfe88fc97676608e074176f90059ec3f992c02

closed 2020-02-24 by jazdw