MangoAutomation/ma-dashboards #303: CRON pattern for history sync is not getting saved on the PTCP publisher.

Joelhaggar, 2019-10-18:
I set the CRON twice and saved and each time I go back to the publisher it is blank and no history syncs are being done.

Craig-IAS, 2019-10-18:
Yes, I also noticed this today, as well as the encryption key. When saved on the new UI do not appear on the old UI

terrypacker, 2019-10-18:
@Joelhaggar I can reproduce this and will look into it, likely a REST model bug.  
@Craig-IAS I cannot reproduce this.  Be more specific what is the "encryption key"  We have  `Shared key` and `Authorization key` as settings.  Either way both of those seem to be fine.  You may need to add screenshots as it may depend on other settings.

terrypacker, 2019-10-18:
[image]
@jazdw it looks like the `syncPattern` field is not being set for the PUT/POST.  I checked the UI code and didn't see why that doesn't work so I'll leave this with you.

Craig-IAS, 2019-10-18:
@jazdw I can't reproduce it now either, I must have been confusing myself earlier.

closed 2019-10-29 by jazdw